davidwernhart/AlDente-Pro-Bug-Tracking #3: After top up battery numbers are not exact anymore

tosbsas, 2021-04-15:
Its wierd

I have set the battery to 80% and it worked perfect there. After using top Up the value is slightly off

[image]

tosbsas, 2021-04-16:
I tested a bit - what's messing up the numbers a bit is the don't charge when sleep setting. With this turned of you messure is exact, with turned off its 1% off - interesting

tosbsas, 2021-04-17:
another test :-)) Turned map off over night and disconnected current. On starting the machine AlDente showed 80% but than went to 78 where it stayed and the symbol went to - done. I re-entered 80 in the max value and it charged to 80 and stayed there.

MatthiasKerbl, 2021-04-19:
I think your MacBook stays below 80% because of the Sailing Mode being activated. You can see that because of the paper plane in the slider. Sailing Mode will keep your MacBook in a charging range and only charge back up to Max. Charge when the lower limit was hit.

closed 2021-04-19 by MatthiasKerbl

tosbsas, 2021-04-19:
Ok. I see. Thanks for the info