davidwernhart/AlDente-Pro-Bug-Tracking #6: Feature request include battery health and other data from ioreg

mikebore, 2021-04-16:
Would be great if all the battery data available from ioreg and other apps (Coconut, Better Battery 2 etc) could be included in AlDente.

MatthiasKerbl, 2021-04-19:
Hi @mikebore,

thanks for your feedback. This is on our list for potential future updates.

julienbdx, 2021-05-14:
Hi!
In the same way, it would be great if AlDente could keep history of battery informations : for example, history of circle count, battery health and max battery capacity in order to track the benefits of using AlDente, history of charge/discharge periods, ...
Thanks

MatthiasKerbl, 2022-11-08:
Hi everyone,

Just letting you know that we are closing this repository since we continued using the original AlDente repository: https://github.com/davidwernhart/AlDente-Charge-Limiter
However, we have not forgotten about your feature request and we have this thread open about it: https://github.com/davidwernhart/AlDente-Charge-Limiter/issues/250

Cheers,
Matthias

closed 2022-11-08 by MatthiasKerbl