davidwernhart/AlDente-Pro-Bug-Tracking #8: Add an option to make "Sailing Mode" always active and not only when the battery is charged to max

SuperGHUB, 2021-05-06:
Hello,

on my Lenovo Laptop it is possible to set a range in which the charging is inhibited, e.g. between 40-80%. So when the state of charge is at 60% and I plug the charger in it stays at 60%, but when the soc is at 35% and I plug in the charger it automatically starts charging to 80% and then stops (Its so smart that it won't stop at 40% when its below the desired minimum, it then charges to the desired max).
In my understanding Sailing Mode is very similar, but it will always charge to the desired maximum and then stops charging between 40-80%(In this example). When I then unplug the charger and use the Laptop for a while and replug the charger at 60%, it will again start charging to 80%.

Is it possible to implement this smart behavior into AlDente? Inhibit charging between 40%-80%, automatically start charging below 40% up to 80% and not only to 40%

At the moment I do this for myself with the slider(when it reaches my max I slide it down to 40% and when I want to charge I slide it back up to 80%), but it would be cool to get this automated.

Thanks in Advance.

MatthiasKerbl, 2021-05-07:
Hi @SuperGHUB,

This sounds like a good upgrade to the Sailing Mode. We will see what we can do about this.

holygrolli, 2021-06-25:
I am very confused now. From the docs/blog I supsected Sailing Mode to work exactly like described above
> Afterwards, the charging of the MacBook is stopped and the MacBook remains at 60% charge level and only charges again when the lower interval limit of 50% has been reached.

I just thought I have a bug in AlDente Pro 1.04. My charge limit is set to 79% with sailing mode 19%. I repeatedly discharged my MacBook to about 65% and was suprised seeing it being charged again to 79%.
> Since it is theoretically healthier for the battery to be charged 10% once instead of 1% 10 times, we have included the Sailing Mode feature

This is what I expected from sailing mode. So can anyone please explain to me, what of **my** scenario differs from the **actual** implementation of Sailing Mode?

Thanks!!

SuperGHUB, 2021-06-25:
Yes, you unplugged your MacBook and thats why the Sailing Mode is Reset and starts again from the beginning. That is also the reason why I'm asking for this special version of Sailing Mode because now it is nearly useless for me.

MatthiasKerbl, 2021-06-26:
> Yes, you unplugged your MacBook and thats why the Sailing Mode is Reset and starts again from the beginning. That is also the reason why I'm asking for this special version of Sailing Mode because now it is nearly useless for me.

Hi @SuperGHUB,
we are working on updating the Sailing Mode with a future update.

noldor-timma, 2021-07-08:
I have pro 1.0.4 and sailing mode does nothing. The charge just stays at max. Behaviour is the same as the free version.

MatthiasKerbl, 2021-07-13:
> I have pro 1.0.4 and sailing mode does nothing. The charge just stays at max. Behaviour is the same as the free version.

Hi @noldor-timma,
did you read our blog post about the Sailing Mode?

holygrolli, 2021-09-01:
> Hi @SuperGHUB,
> we are working on updating the Sailing Mode with a future update.

Any news on this?

rimusz, 2021-09-01:
> Any news on this?

+1

rimusz, 2022-02-09:
@MatthiasKerbl any news on this? 
It is almost one year it was asked for as paid customers :)

MatthiasKerbl, 2022-02-09:
> @MatthiasKerbl any news on this? It is almost one year it was asked for as paid customers :)

Hi @rimusz,

We included the suggestions of @SuperGHUB already a couple of months ago since AlDente Pro 1.08. :)

Cheers,
Matthias

closed 2022-02-09 by MatthiasKerbl