davidwernhart/AlDente-Pro-Bug-Tracking #9: The sense of using AlDente on MacBook in clamshell mode

JhonnyAxe, 2021-05-16:
Hi there! I'm wondering if there is any sense of using and even buying AlDente Pro on my MacBook which, generally, I use like a desktop computer (it's connected with a charger and external display). I've seen in the FAQ that the "Discharge" function isn't supported for that case. 

Thus, I suppose, that the program won't do its job because the charge percentage is always that which I set (70%/80%/100%). 

Moreover, other features of PRO like “Sailing Mode,” “Heat Protection” and “Calibration Mode” are useless because "Discharge" is not supported for clamshell mode.

Am I right?

davidwernhart, 2021-05-16:
Hi @JhonnyAxe 

All features involving active discharging (Discharge and Calibration Mode) do not work when your MacBook is plugged in and the lid is closed. This is due to limitations with Apple's implementation of the low-level system calls we utilize to make these features work.

This being said, charge limiting, the primary use case of AlDente is still functional in clamshell mode. If you start at 50% for example, set your charge limit to 70% and use your notebook in clamshell mode, it should stop charging and stay at 70% as expected.
Both Sailing Mode and Heat Protection do also work as described on our website, even when the MacBook is closed and connected to an external display.

Hope that clears things up for you.

Best regards,
David

JhonnyAxe, 2021-05-16:
Thanks, I'm gonna try it

closed 2021-05-16 by JhonnyAxe