davidwernhart/AlDente-Pro-Bug-Tracking #11: Sailing mode always keeps battery at charge limit

jcass77, 2021-05-29:
I have configured a charge limit of 80% and a sailing mode difference of 15%. The 'Optimize battery charging' OSX setting is also disabled.

My understanding is that these settings should cause the battery to charge to 80%, discharge to 65%, and then repeat this charging cycle indefinitely?

Instead, AlDente charges the battery to 80% and then just keeps it charged at exactly that level forever.

MatthiasKerbl, 2021-05-29:
> I have configured a charge limit of 80% and a sailing mode difference of 15%. The 'Optimize battery charging' OSX setting is also disabled.
>
> My understanding is that these settings should cause the [...]

Hi @jcass77,

The Sailing Mode does not charge and discharge your MacBook in a certain range. This would result in a faster battery degradation over time. You can read about what the Sailing Mode actually does in our blog article: https://apphousekitchen.com/feature-explanation-sailing-mode/

Kind regards,

Matthias

jcass77, 2021-05-29:
Thanks for the feedback. I’m afraid I still don’t get it, the block article states:

> The current charge level of the MacBook is 80% and the MacBook is plugged in and charging is paused because the charge limit on AlDente Pro is set to 80%. Sailing Mode is activated and an interval of 5% is set.The MacBook will not charge again until the lower limit of the interval, 75%, is reached.

How is that different to the scenario that I described above?

MatthiasKerbl, 2021-05-29:
The difference is how the battery gets to the lower percentage of the interval. When the Sailing Mode is activated, the MacBook will not "actively" discharge the battery to the lower percentage. However, the battery will still lose some energy over time and AlDente will only start charging again, when the lower percentage has been reached.

jcass77, 2021-05-29:
Ok, but how will the battery "still lose some energy over time" if the laptop is connected to power?

The tooltip for the sailing mode setting states:

[image]

...which makes it sound like AlDente is supposed to let the battery drain to: `charge limit - sailing mode difference` before charging again.

Not sure if that would actually be desirable from a battery longevity point of view - just trying to figure out what sailing mode actually does.

jcass77, 2021-05-29:
Maybe there are some technical details to how battery charging on the MacBook works that I am not aware of.

Is the following accurate?

1. The battery charges to 'charge limit' and then switches over to the power supply

2. The battery will then only discharge if (a) a brief surge in power requirements causes the MacBook to use the battery as a buffer (I have never observed this), or (b) the battery slowly drains over a period of weeks or months while the laptop remains plugged in (but not actively running on the battery).

3. Whereas the MacBook would normally start charging the battery to 100% again as soon as 2(a) or 2(b) occurs, sailing mode will allow the battery to discharge to `charge limit - sailing mode difference` again before charging.

It sounds like the above might be fairly rare, and that most users will never notice the effect of sailing mode (even though it is doing its thing)?

MatthiasKerbl, 2021-05-29:
> Maybe there are some technical details to how battery charging on the MacBook works that I am not aware of.
>
> Is the following accurate?
>
> 1. The battery charges to 'charge limit' and then switches [...]

1. correct
2. a)b) correct
3. correct

You are right, most users will not notice the Sailing Mode but many users wanted a feature like that and I like to use it as well.

I hope you understand the Sailing Mode in full detail now. :)

MatthiasKerbl, 2021-05-29:
> Ok, but how will the battery "still lose some energy over time" if the laptop is connected to power?
>
> The tooltip for the sailing mode setting states:
>
> [image]
>
> ...which makes it sound like AlDente [...]

I see where the confusion comes from. Maybe we should change this text to avoid this confusion.

closed 2021-06-01 by MatthiasKerbl