davidwernhart/AlDente-Pro-Bug-Tracking #13: License problems @ v.1.04

niklashaug, 2021-06-14:
After updating to v.1.04 I had to manually activate my license again to get the pro version.

davidwernhart, 2021-06-14:
Hi Niklas,

thank you for the feedback!
AlDente 1.04 indeed has undergone some major changes regarding licensing but this never happened in testing.
Have you been disconnected from the internet while trying to activate the app?

Also, could you maybe shoot us an email at support@apphousekitchen.com? This way, I am able to check your license

Best regards,
David

niklashaug, 2021-06-14:
I don't think I was disconnected from the internet as I was able to download the update without any problems.
Just sent you an email.

AaronAkane, 2021-06-14:
having same issue here showing updated features but is not keeping my licence for pro version. Keeps asking me to download pro or get pro version. Even when using email ink sent with licence code and download doesn't work, will keep telling me I don't have a pro licence. Which I have in my email my licence code but no where to enter it or even the window doesn't come up asking for your licence number when installing pro version and updating. Seems just to be lost as it is not a connection issue with wifi or data or different wifi connection tried all three seeing if it might be a signal to strength issue in wifi or data receiving. It is not so please address this issue as I have paid for pro version but have no access to any of the features after update even going back to previous version does not solve the issue

MatthiasKerbl, 2021-06-14:
> having same issue here showing updated features but is not keeping my licence for pro version. Keeps asking me to download pro or get pro version. Even when using email ink sent with licence code and download doesn't work, will keep telling me I don't have a pro licence. Which I have in my email my licence code but no where to enter it or even the window doesn't come up asking for your licence number when installing pro version and updating. Seems just to be lost as it is not a connection issue with wifi or data or different wifi connection tried all three seeing if it might be a signal to strength issue in wifi or data receiving. It is not so please address this issue as I have paid for pro version but have no access to any of the features after update even going back to previous version does not solve the issue. All pro features are now disabled even if I have a licence key which in never asks for or can't even find an option to put in

Hi @AaronAkane,

we are really sorry for the inconvenience. Please install the newest version (1.04) and go into settings->General Settings->Manage License, put in your email address and your license code. Afterwards, everything should work as intended.

Kind regards,

Matthias

alfattal, 2021-06-18:
I'm also having the same problem after updating. I've uninstalled the Pro version but don't have the link to download and install the latest version. I've sent an email to the support but no response so far. Can you please help me with this?

closed 2021-06-23 by davidwernhart