davidwernhart/AlDente-Pro-Bug-Tracking #15: Charging to 100% when lid closed

Brewforce, 2021-06-23:
Hi, when I close my laptop (MacBook Air M1, Big Sur 11.4) the charge don't stop and fill the battery to 100%.
I've AlDente Pro, can you take a look at this ?

Best Regards

davidwernhart, 2021-06-23:
Hi @Brewforce 

do you have "Stop Charging when Sleeping" enabled?

Best regards,
David

Brewforce, 2021-06-23:
No I haven't. I have to in order to keep the battery under de limit I've decided ?

davidwernhart, 2021-06-23:
Yes, you have to.
AlDente cannot operate in sleep mode, since macOS does not allocate CPU time to 3rd party apps when sleeping.
Because of that, AlDente is not able to actively monitor your battery status and switch off charging at the right time when you close the MacBook.
The only thing we are able to do is detect when the system is about to go to sleep and stop charging at the current percentage beforehand.

Hope that clears things up,
David

Brewforce, 2021-06-23:
Yes it does thank you very much !

davidwernhart, 2021-06-23:
Glad I was able to help! :)

closed 2021-06-23 by davidwernhart