davidwernhart/AlDente-Pro-Bug-Tracking #17: Correct charge level not displayed in Monterey 12.0-beta (21A5284e)

napthemax, 2021-07-24:
I know I'm running Monterey beta, but could still be valid to keep an eye on for you. AlDente-Pro does not show correct SoC of my computer. It freezes on the level that it have on reboot. See screenshot comparing with CoconutBattery.

[image]

MatthiasKerbl, 2021-07-25:
> I know I'm running Monterey beta, but could still be valid to keep an eye on for you. AlDente-Pro does not show correct SoC of my computer. It freezes on the level that it have on reboot. See screenshot comparing with CoconutBattery.
> 
> [image]

Hi @napthemax,
Try to reinstall AlDente Pro after the update to the macOS Monterey.

napthemax, 2021-07-25:
Thank you! That did the trick :-)

closed 2021-07-25 by napthemax