davidwernhart/AlDente-Pro-Bug-Tracking #18: Can't modify 'difference' value for sailing mode

swiftgaruda, 2021-08-09:
Hi, I'm running Big Sur 11.4 (about to update to 11.5.1) on a Macbook Pro 2019 15" (2.3 i9) and I can't seem to modify the difference value of the sailing mode. I had it set to 35% difference, which should sail the battery down to 25%, but now I'm not able to modify that value at all. 

Should I try resetting the SMC?

Thanks in advance, I LOVE AlDente. Can't believe I didn't find it earlier....

https://user-images.githubusercontent.com/16741392/128710452-31b57a2b-4155-49e7-a273-a0008762bb5c.mov

MatthiasKerbl, 2021-08-19:
Hi @swiftgaruda,

I think the percentages you chose are too big. Try a value between 5-30%.

swiftgaruda, 2021-08-20:
Hi Mattias, thanks for your reply. I actually am unable to change the
values at all—that's the original bug that I'm reporting. Still in the same
situation.

Thanks,
Daniel

MatthiasKerbl, 2021-08-24:
Have you tried to completely reinstall AlDente Pro again?