davidwernhart/AlDente-Pro-Bug-Tracking #19: 100% CPU usage

RobinDaugherty, 2021-08-15:
Running version 1.06
macOS 11.5.1
MacBook Pro (13-inch, M1, 2020)

Seems to happen only when on battery. After quitting and reopening, it continues.

badwulfy, 2021-08-17:
I confirm I have the same issue with the version 1.06 from SetApp.
I use macOS beta 12.0 (21A5304g).
Several times a day, Al Dente use more than 100% CPU without interruption.

dotqwerty, 2021-08-18:
Same issue here. This is a bad bug.
[image]

davidwernhart, 2021-08-18:
Hi Guys!

We are aware of this bug and have finally been able to replicate it.
A fix will be released soon!

Thank you for your patience,

David
AppHouseKitchen

badwulfy, 2021-08-18:
Thank you very much for your help. Take the time you need.

In the meantime, I have disabled Al Dente Pro and re-enabled Apple's battery optimization system.
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

falcn, 2021-08-19:
>  In the meantime, I have disabled Al Dente Pro and re-enabled Apple's battery optimization system.
I only disable AlDente when I'm on battery. It works correctly when connected to charger.

falcn, 2021-08-19:
I can confirm that this issue is fixed in 1.07

davidwernhart, 2021-08-19:
Glad the fix worked for you! Best regards, David

closed 2021-08-19 by davidwernhart

RobinDaugherty, 2021-08-24:
🌮 🎉 🥇