davidwernhart/AlDente-Pro-Bug-Tracking #22: Heat protection incorrectly saves user-set value

falcn, 2021-08-18:
I have MacOS Monterey beta (21A5294g) running on MacBook Air M1 2020.
AlDente Pro 1.06 from Setapp.

Steps to reproduce:
1) open settings, enable Heat protection. I have 3° as Max Temp from earlier bugged save.
[image]

2) change "Max Temp" to 36, close settings
3) Heat protection activates, because AlDente uses 3° instead of user-set 36°
[image]

4) Close AlDente, reopen Settings. Heat protection is set to 3°
[image]

MatthiasKerbl, 2021-08-22:
Hi @falcn,

You have to press enter after entering a new value in Heat Protection.
I can only reproduce your issue if I only change the value without pressing enter afterward.

falcn, 2021-08-22:
I pressed "Enter", it still inhibits charging. 
Restarting Aldente resets visible value to 3°.

[image]

falcn, 2021-08-22:
Can you tell me where config file is located? I'll try changing it manually

falcn, 2021-08-22:
By the way, same thing happens to "Sailing mode", it just defaults to 5% after restart no matter what value was entered.

MatthiasKerbl, 2021-08-25:
Hi @falcn,

Have you tried to completely reinstall AlDente Pro and update to our newest version 1.07?

falcn, 2021-11-29:
I don't know about 1.07, but 1.10 works. Thank you

closed 2021-11-29 by falcn