davidwernhart/AlDente-Pro-Bug-Tracking #23: AlDente indicates charging, but does not actually charge.

407pilot, 2021-08-21:
AlDente Pro 1.07
MacOS Big Sur 11.5.2

Occasionally AlDente Pro will indicate that it is charging, but will not actually charge. Heat control is enabled, but battery temp is reported below threshold. Have to unplug, then plug in, the power cord to get the system to charge.

MatthiasKerbl, 2021-08-23:
Hi @407pilot,

Which MacBook model do you use?
Usually, there is a delay of up to 1-2 minutes for changes to be displayed by macOS. Are you sure it doesn’t start to charge at all?

407pilot, 2021-08-23:
It's a 2019 MacBook Pro. 

Yes, very sure. It happened on Friday and I let it go until the battery got down to 5%; well over an hour of use. It wasn't only not charging the battery, it wasn't using power to run the computer. I verified this using iStat Menus.

407pilot, 2021-08-23:
I'm positive that discharge wasn't selected. The menu bar icon shows power is applied and charging is paused when it happens.

MatthiasKerbl, 2021-08-24:
Have you tried to completely reinstall AlDente Pro?

closed 2022-03-10 by MatthiasKerbl