davidwernhart/AlDente-Pro-Bug-Tracking #24: Feature Request - Add fan control to heat protection

407pilot, 2021-08-21:
In order to better manage heat, add the ability to override the system and increase fan speed in anticipation of temperature rise while charging.

MatthiasKerbl, 2021-08-23:
Hi @407pilot,

Sounds like a good addition to our Heat Protection feature. We will look into this.

Cheers,
Matthias

MatthiasKerbl, 2022-11-08:
Hi @407pilot,

Just letting you know that we are closing this repository since we continued using the original AlDente repository: https://github.com/davidwernhart/AlDente-Charge-Limiter
However, we have not forgotten about your feature request and we have this thread open about it: https://github.com/davidwernhart/AlDente-Charge-Limiter/issues/630

Cheers,
Matthias

closed 2022-11-08 by MatthiasKerbl