davidwernhart/AlDente-Pro-Bug-Tracking #25: AlDente Pro messing with my sleep prevention settings

zahlex, 2021-08-22:
Thank you so much guys for providing this amazing piece of software 👍
I'd like to report a convince issue with the lastest update:

I did set my Apple Silicon MacBook to true always on mode via `sudo pmset -a disablesleep 1`, so I can close the lid and carry it around without interrupting any tasks, media playback, etc.

But since the latest update of AlDente Pro, every time I plug the charging cable in or out, this setting is set back to default and I have to execute the command again. It would be nice if you would find a way around this. Thank you in advance 🙂

MacBook Pro (13-inch, M1, 2020)
macOS Big Sur 11.5.2 (20G95)
AlDente Pro 1.07

MatthiasKerbl, 2021-08-24:
Hi @zahlex,

I am glad you like our little piece of software.
As a safety feature, AlDente resets the sleep setting of your MacBook. Therefore, we can not deactivate it for everyone. However, if we see a demand of a setting to disable this safety feature we could include it in a future update.

Cheers,
Matthias

zahlex, 2021-08-31:
Hello @MatthiasKerbl,

thank you very much for your fast response.

Older versions didn't have this issue, so how was it handled back then? Couldn't this be solved by a functionality that detects if there are some user issued changes and reset to those instead of system default?

Kind Regards,
Alex

MatthiasKerbl, 2021-09-19:
Older versions of AlDente didn't control the sleeping behavior of MacBooks. Therefore, no safety feature was needed. Your suggestion might be a possibility. We will look into this.

Cheers,
Matthias

closed 2022-03-01 by MatthiasKerbl