davidwernhart/AlDente-Pro-Bug-Tracking #28: Sailing mode with discharge option

rimusz, 2022-02-10:
@MatthiasKerbl 
It would be really nice to support in `Sailing mode` discharge as well.
E.g. I set in there `Difference: 20%` so it will discharge to `60%` and will charge back to `80%`
That would really allow the battery to get discharged.

MatthiasKerbl, 2022-02-10:
Hi @rimusz,

We could implement a feature like that. However, we decided not to do it because battery research shows that keeping your battery at a healthy percentage is much better for long-term battery health than cycling through the battery unnecessarily. This would only result in a faster increase of charge cycles and therefore result in a faster battery degradation which we try to avoid with AlDente.

Cheers,
Matthias

rimusz, 2022-02-10:
ok, if your really think the current implementation is the best way to safe the battery life, sure :)

closed 2022-02-10 by rimusz